Xinyuan-LilyGO/LILYGO-T-display-RP2040 #2: Unable to get the TFT to work with Micro Python

Nimaear, 2021-12-13:
I have been unable to get the display to work with micropython. I have looked around for different implementations, but seem unable to find one that works.

Can you point me to a library that works with micropython?

slabua, 2021-12-27:
@Nimaear try this https://github.com/russhughes/st7789py_mpy
I have not checked yet, but hopefully I will soon.

Nimaear, 2021-12-27:
I tried this both of next using this library without any luck. Not sure what I am doing wrong here.

```python
def main():
    spi = SoftSPI(
        baudrate=20000000,
        polarity=1,
        phase=0,
        sck=Pin(2),
        mosi=Pin(3),
        miso=Pin(6))

    tft = st7789.ST7789(
        spi,
        135,
        240,
        reset=Pin(0, Pin.OUT),
        cs=Pin(5, Pin.OUT),
        dc=Pin(1, Pin.OUT),
        backlight=Pin(4, Pin.OUT),
        rotation=0)
```

and 

```python
def main():
    spi = SPI(
        baudrate=20000000,
        polarity=1,
        phase=0,
        sck=Pin(2),
        mosi=Pin(3),
        miso=None)

    tft = st7789.ST7789(
        spi,
        135,
        240,
        reset=Pin(0, Pin.OUT),
        cs=Pin(5, Pin.OUT),
        dc=Pin(1, Pin.OUT),
        backlight=Pin(4, Pin.OUT),
        rotation=0)
```

slabua, 2021-12-30:
@Nimaear try this! The display needs to be explicitly powered on with the first 2 lines:

```
    p22 = Pin(22, Pin.OUT)
    p22.value(1)

    spi = SoftSPI(
        baudrate=20000000,
        polarity=1,
        phase=0,
        sck=Pin(2),
        mosi=Pin(3),
        miso=Pin(13))

    tft = st7789.ST7789(
        spi,
        135,
        240,
        reset=Pin(0, Pin.OUT),
        cs=Pin(5, Pin.OUT),
        dc=Pin(1, Pin.OUT),
        backlight=Pin(4, Pin.OUT),
        rotation=1)
```

erongd, 2022-02-09:
@slabua I tried your code, it does not seem to work. Only the backlight turned on. No other operation works.

slabua, 2022-02-09:
@erongd have you tried pulling the latest from https://github.com/russhughes/st7789py_mpy ? Russ merged my PR with the example. If it still doesn't work, I'm not sure why :\

erongd, 2022-02-10:
@slabua I did not realize you're using the python version of the driver. I am trying to use the C version of the driver. Will try again with the python version.

russhughes, 2022-02-11:
This works with the C based driver. 

```
from machine import Pin, SPI
import st7789

Pin(22, Pin.OUT, value=1)

spi = SPI(
    0,
    baudrate=40000000,
    polarity=1,
    phase=0,
    sck=Pin(2, Pin.OUT),
    mosi=Pin(3, Pin.OUT),
    miso=None)

tft = st7789.ST7789(
      spi,
      135,
      240,
      cs=Pin(5, Pin.OUT),
      dc=Pin(1, Pin.OUT),
      backlight=Pin(4, Pin.OUT))

tft.init()
```

LilyGO, 2022-03-17:
Looks like the problem has been solved, open it again if necessary

closed 2022-03-17 by LilyGO