Xinyuan-LilyGO/LILYGO-T-display-RP2040 #4: Cannot compile with TFT_eSPI

markjuggles, 2021-12-30:
I have wasted many hours trying to get the TFT_eSPI library to compile but there just isn't enough information.  How was this done?  

https://github.com/Bodmer/TFT_eSPI.git

It looks like my LilyGO boards are going to the landfill.

LeFauve, 2022-01-28:
I hope your boards aren't in landfill yet because I just made a pull request to add a user setup file for the LilyGo T-Display RP2040 in Bodmer's repository.

Once it's merged, just edit the User_Setup_Select.h to select "User_Setups/Setup137_LilyGo_TDisplay_RP2040.h" and comment out any #define XXX_DRIVER line in User_Setup.h (for some reason one of those lines is uncommented by default, and it creates havoc on the screen).

markjuggles, 2022-01-28:
No, I made a 2nd attempt to use them and failed.  All of my Ali Express
emails are getting deleted without being opened until this is resolved
though. If you have a fix, I will be happy to try it out. A working RP2040
with display would be very useful - if the display actually works.

Thank you

LeFauve, 2022-01-29:
You're welcome :o)

My pull request has been merged in TFT_eSPI so you may give it a try.
I created another pull request for this repository to update the Arduino IDE instructions with full procedure to install and configure TFT_eSPI lib. Until it's accepted, you may [look into my branch](https://github.com/LeFauve/LILYGO-T-display-RP2040/tree/Update-Arduino-instructions) (only the readme has been modified, all other files are the same than the main repository ones).

markjuggles, 2022-01-30:
You.
Are.
AWESOME!

The "firmware" project works out-of-the-box as I would expect a demo to do.

I then attempted Cellular Automata since I am a fan of the "Game of Life".
That was a little more challenging.

Your text is:

<snip>

Remarks:

   - You can also use other sample programs provided by TFT_eSPI with some
   minor changes (mostly changing the screen size and switching the backlight
   on with something like the following line)

analogWrite(4, 255); // pin 4 controls the backlight; values between 0
(off) and 255 (full brightness) are accepted

</snip>

I would change that to "Copy most of the LilyGo firmware setup() into the
sample program setup() with the required #includes and #defines.  Comment
out the pushImage() call.  Set the hard-coded screen sizes to TFT_HEIGHT
and TFT_WIDTH which are defined correctly.

THANK YOU!

LeFauve, 2022-01-30:
Funny you did this because I also played with the "Game of Life" example last night :o)
I realized I should make an update because you need not only to turn backlight on but more important, you need to power up the screen using pin 22. 
Also I made a small mistake on how the lib is supposed to be used: Instead of editing User_Setup.h, you should just not include it from User_Setup_Select.h.

I made a lot of changes to the Game of Life if you're interested (fix the "wrapping" bug, add the possibility to restart with a button, from either random or a glider gun (looks amazing), colors (newborn cells are green, just killed ones are red), and a lot of code optimization).
I'm not planning on pushing my version to TFT_eSPI because it's kind of RP2040 specific now (and I'm planning to try using both cores for better performances) but I suppose I could put it on GitHub.

A word of advice about this LilyGO board though. The two buttons near the USB plug are not as sturdy as they look. One of mine broke after only about 50 pushes...

alleebell, 2022-03-01:
If anyone has problems getting the display of LilyGO T-Display RP2040 to work. Please read all the postings above carefully. 
Thanks LeFauve and markjuggles!

I ended up adding the following to turn ON the backlit as well as the TFT.
  pinMode(4, OUTPUT);
  digitalWrite(4, HIGH); // Backlight on
  pinMode(22, OUTPUT);
  digitalWrite(22, HIGH); // Turn on module

My next challenge is to use the I2C to read temperature sensors such as AHT-20! 
Any suggestions!

LilyGO, 2022-03-17:
Sorry for not responding to your question, all the pins used on the board have been shown in the schematic and the newly uploaded factory example

closed 2022-03-17 by LilyGO

RC0D3, 2024-09-16:
> If anyone has problems getting the display of LilyGO T-Display RP2040 to work. Please read all the postings above carefully. Thanks LeFauve and markjuggles!
>
> I ended up adding the following to turn [...]

Ty very much, worked fine!