Xinyuan-LilyGO/LILYGO-T-display-RP2040 #5: Need USB-PID for Circuitpython support

erongd, 2022-03-20:
I am working on add circuitpython support for this board. Would appreciate help with obtain official USB-PID for CircuitPython and UF2-Bootloader. Please refer to pull request https://github.com/adafruit/circuitpython/pull/6037.

LilyGO, 2022-03-29:
We are willing to add USB-PID for this board. Thank you for the information you provided

erongd, 2022-03-29:
Thanks for the reply, please let me know the proper VID/PID to use for circuitpython. I will update the code accordingly so it can be merged into official cpy.

fabaff, 2022-05-19:
@LilyGO, any progress with the PID?  Thanks for feedback.

Looks like that the board is still not listed on https://github.com/raspberrypi/usb-pid.

leeroywking, 2022-05-20:
+1 to this issue, the examples for micropython are unusable and the circuit python support would be greatly appreciated

telemac73, 2022-05-24:
+1 we realy need circuitpython Thanks

LilyGO, 2022-05-24:
I've already submitted an application, but I don't seem to have received any feedback. Is there any other way to apply?

fabaff, 2022-08-26:
Write an email to [info@raspberrypi.com](mailto:info@raspberrypi.com) could help. The process to request a PID is no very straight forward..

fabaff, 2022-12-10:
The request for the PID has to come from the OEM manufacturer and @LilyGO made on a couple of month ago. If @raspberrypi is not processing it then there is not much we can do.

kreier, 2023-03-08:
> We are willing to add USB-PID for this board. Thank you for the information you provided

> I've already submitted an application, but I don't seem to have received any feedback. Is there any other way to apply?

Hi Liligo, can you fill out another request on https://github.com/raspberrypi/usb-pid for the T-Display rp2040 and another one for the T-PicoC3 please? Erongd created a working pull request with correct pin description and driver integration almost a year ago. And it's posted on the forum as well:

https://community.lilygo.cc/topic/50/help-with-usb-vid-pid-for-circuitpython

Thank you!