Xinyuan-LilyGO/LILYGO-T-display-RP2040 #7: Problems Compiling Arduino Example

MadGeometer, 2022-05-06:
Hello,

I'm unable to compile the file example/Arduino/firmware/firmware.ino.

There's a syntax error at line 64 in the file  ,and after commenting that line, am still unable to compile. Error messages pasted below.

Any help would be greatly appreciated!

Thanks,
Mike

Arduino IDE version: 2.0.0-rc6  or 1.8.19
TFT_eSPI version 2.4.51
In User_Setup_Select.h...
 - The line "#include <User_Setups/Setup135_ST7789.h>" has been un-commented 
 - The line "#include <User_Setup.h>" has been commented

Here's the output:
In file included from /Users/mikeklepper/Google Drive/RP2040/T-Display/LILYGO-T-display-RP2040-main/example/Arduino/firmware/firmware.ino:1:
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:875:6: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  875 |     #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |      ^~~~~~~
In file included from /Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:875:6: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  875 |     #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |      ^~~~~~~
In file included from /Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:26:
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.c:18:62: error: 'TFT_SCLK' was not declared in this scope
   18 |     SPIClassRP2040 spi = SPIClassRP2040(SPI_X, TFT_MISO, -1, TFT_SCLK, TFT_MOSI);
      |                                                              ^~~~~~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.c:18:72: error: 'TFT_MOSI' was not declared in this scope; did you mean 'TFT_MISO'?
   18 |     SPIClassRP2040 spi = SPIClassRP2040(SPI_X, TFT_MISO, -1, TFT_SCLK, TFT_MOSI);
      |                                                                        ^~~~~~~~
      |                                                                        TFT_MISO
In file included from /Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:74,
                 from /Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:155:44: warning: left shift count is negative [-Wshift-count-negative]
  155 |       #define CS_L sio_hw->gpio_clr = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:75:5: note: in expansion of macro 'CS_L'
   75 |     CS_L;
      |     ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::begin_nin_write()':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:155:44: warning: left shift count is negative [-Wshift-count-negative]
  155 |       #define CS_L sio_hw->gpio_clr = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:87:5: note: in expansion of macro 'CS_L'
   87 |     CS_L;
      |     ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_write()':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:156:44: warning: left shift count is negative [-Wshift-count-negative]
  156 |       #define CS_H sio_hw->gpio_set = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:101:7: note: in expansion of macro 'CS_H'
  101 |       CS_H;
      |       ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::end_nin_write()':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:156:44: warning: left shift count is negative [-Wshift-count-negative]
  156 |       #define CS_H sio_hw->gpio_set = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:116:7: note: in expansion of macro 'CS_H'
  116 |       CS_H;
      |       ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:155:44: warning: left shift count is negative [-Wshift-count-negative]
  155 |       #define CS_L sio_hw->gpio_clr = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:136:5: note: in expansion of macro 'CS_L'
  136 |     CS_L;
      |     ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_read()':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:156:44: warning: left shift count is negative [-Wshift-count-negative]
  156 |       #define CS_H sio_hw->gpio_set = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:157:7: note: in expansion of macro 'CS_H'
  157 |       CS_H;
      |       ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writedata(uint8_t)':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:155:44: warning: left shift count is negative [-Wshift-count-negative]
  155 |       #define CS_L sio_hw->gpio_clr = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:958:3: note: in expansion of macro 'CS_L'
  958 |   CS_L;        // Allow more hold time for low VDI rail
      |   ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'uint8_t TFT_eSPI::readcommand8(uint8_t, uint8_t)':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:156:44: warning: left shift count is negative [-Wshift-count-negative]
  156 |       #define CS_H sio_hw->gpio_set = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:994:3: note: in expansion of macro 'CS_H'
  994 |   CS_H; // Some displays seem to need CS to be pulsed here, or is just a delay needed?
      |   ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:155:44: warning: left shift count is negative [-Wshift-count-negative]
  155 |       #define CS_L sio_hw->gpio_clr = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:995:3: note: in expansion of macro 'CS_L'
  995 |   CS_L;
      |   ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual uint16_t TFT_eSPI::readPixel(int32_t, int32_t)':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:156:44: warning: left shift count is negative [-Wshift-count-negative]
  156 |       #define CS_H sio_hw->gpio_set = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:1147:3: note: in expansion of macro 'CS_H'
 1147 |   CS_H;
      |   ^~~~
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::readRectRGB(int32_t, int32_t, int32_t, int32_t, uint8_t*)':
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h:156:44: warning: left shift count is negative [-Wshift-count-negative]
  156 |       #define CS_H sio_hw->gpio_set = (1ul << TFT_CS)
/Users/mikeklepper/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:2066:3: note: in expansion of macro 'CS_H'
 2066 |   CS_H;
      |   ^~~~

Compilation error: exit status 1

numeru55, 2022-05-10:
Using my Mac, build and run are successful. Please try.

- Arduino 1.8.19
- Board: Raspberry Pi Pico Earle F. Phihower, III 2.0.0
- Library: TFT_eSPI by Bodmer 2.4.51

uncomment at `~/Documents/Arduino/libraries/TFT_eSPI/User_Setup_Select.h`
```~/Documents/Arduino/libraries/TFT_eSPI/User_Setup_Select.h
#include <User_Setups/Setup25_TTGO_T_Display.h>    // Setup file for ESP32 and TTGO T-Display ST7789V SPI bus TFT
``` 
 
uncomment at `~/Documents/Arduino/libraries/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h`
```~/Documents/Arduino/libraries/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h
 #define TFT_BL 4
``` 

comment at `~/Documents/Arduino/libraries/TFT_eSPI/User_Setup.h`
```~/Documents/Arduino/libraries/TFT_eSPI/User_Setup.h
//#define ILI9341_DRIVER       // Generic driver for common displays
```

numeru55, 2022-05-10:
Please try. My Mac could build and run.

- Arduino 1.8.19
- Board: Raspberry Pi Pico Earle F. Phihower, III 2.0.0
- Library: TFT_eSPI by Bodmer 2.4.51

uncomment at `~/Documents/Arduino/libraries/TFT_eSPI/User_Setup_Select.h`
```~/Documents/Arduino/libraries/TFT_eSPI/User_Setup_Select.h
#include <User_Setups/Setup137_LilyGo_TDisplay_RP2040.h>  // Setup file for Lilygo T-Display RP2040 (ST7789 on SPI bus with 135x240 TFT)
``` 
 
uncomment at `~/Documents/Arduino/libraries/TFT_eSPI/User_Setups/Setup137_LilyGo_TDisplay_RP2040.h`
```~/Documents/Arduino/libraries/TFT_eSPI/User_Setups/Setup137_LilyGo_TDisplay_RP2040.h
 #define TFT_BL 4
``` 

comment at `~/Documents/Arduino/libraries/TFT_eSPI/User_Setup.h`
```~/Documents/Arduino/libraries/TFT_eSPI/User_Setup.h
//#define ILI9341_DRIVER       // Generic driver for common displays
```

MadGeometer, 2022-05-12:
Works great! Thanks numeru55 and LilyGO!

closed 2022-05-12 by MadGeometer