Xinyuan-LilyGO/LILYGO-T-display-RP2040 #10: Arduino IDE

tony1tf, 2022-08-23:
In your QuickStart info for adding the Pico board from the earlephilhower board manager did you mean 'Install' not 'Add'
Where is the TFT_eSPI library? I do not understand your explanation of how to install it in the Arduino IDE. Also, I cannot find the Lilygo RP2040 board - do I just select the generic RP2040 board?
Also, I cannot see any serial port for the board on my MacBookPro. Do I need to download a driver?

LilyGO, 2022-08-24:
1. Please search and add the TFT_eSPI library on Arduino.
2. There is no port on the mac book: for the first time, please press and hold the BOOT button and then click reset or connect to the USB. At this point the computer should eject a new disk. Drag and drop your compiled .uf2 file here to complete the firmware upload. If the USB CDC function in Arduino is used. The next time you upload the firmware, you can select the port for uploading.

tony1tf, 2022-08-24:
Hi
Thanks for your help. I have downloaded the TFT library, and tried to compile the 'Sprite_TFT_Rainbow' test program. It complained about some missing items - mainly some touch definitions. Here's the Arduino error messages:

Arduino: 1.8.13 (Mac OS X), Board: "Raspberry Pi Pico, 2MB (no FS), 133 MHz, Small (-Os) (standard), Disabled, Disabled, Disabled, Disabled, None, Pico SDK, IPv4 Only"

In file included from /Users/Tony/Documents/Arduino/libraries/TFT_eSPI/examples/Sprite/Sprite_TFT_Rainbow/Sprite_TFT_Rainbow.ino:22:
/Users/Tony/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  882 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from /Users/Tony/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/Users/Tony/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  882 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from /Users/Tony/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:32:
/Users/Tony/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.c:18:62: error: 'TFT_SCLK' was not declared in this scope; did you mean 'TFT_CS'?
   18 |     SPIClassRP2040 spi = SPIClassRP2040(SPI_X, TFT_MISO, -1, TFT_SCLK, TFT_MOSI);
      |                                                              ^~~~~~~~
      |                                                              TFT_CS
/Users/Tony/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.c:18:72: error: 'TFT_MOSI' was not declared in this scope; did you mean 'TFT_MISO'?
   18 |     SPIClassRP2040 spi = SPIClassRP2040(SPI_X, TFT_MISO, -1, TFT_SCLK, TFT_MOSI);
      |                                                                        ^~~~~~~~
      |                                                                        TFT_MISO
exit status 1
Error compiling for board Raspberry Pi Pico.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

LilyGO, 2022-08-25:
The TFT-eSPI library needs to modify the configuration file to adapt to different boards. The monitor screen driver is ST7789. Have you modified the configuration file.

tony1tf, 2022-08-25:
Hi LilyGO
I have now modified the config file 'User_set_up_select.h' - uncommented the line 'setup137_lilygo......', and commented out the default set up.
Could you explain why this can't be done from the Arduino menu, please.
Anyway - I can now compile the test code to produce a UF2 file. It produces one warning about there being no touch pin defined.  However, I cannot get the board to act as a mass memory - plugging it in to a MacBook Pro, or an old W10 laptop with the boot button pressed. If I do the same with a standard Pico from R Pi, I get the correct mass storage device on both computers.
So at the moment, I am stuck - unable to load any new code into the board.

LilyGO, 2022-08-26:
1. The method of modifying the configuration file "User_set_up_select.h" is decided by the author of TFT_eSPI. We as users can only abide by his regulations.
2. There are two ways to generate a large-capacity device: a. Press and hold the BOOT button and click reset or insert a USB. b, do not modify the USB port settings when programming with Arduino, my configuration diagram is as follows.[image]
[image]

3. T-Display-rp2040 does not have the front and rear USB-C design like T-PicoC3.

The easiest way to verify you can use the micropython firmware provided by us to drag and drop to the T-Display-RP2040. Used to verify whether the flash is good.

tony1tf, 2022-08-26:
Thank you, Lilygo
I have set up a new version of Arduino IDE on a laptop PC running W10. Boards manager downloaded just as your setup.
If I plug in an R Pi Pico, I see it at com3 in 'device manager and in the Arduino SDK.
When I plug in the LilyGO board, it is not recognised by the PC - no serial and no mass memory if I press the boot select button. I have two devices from BangGood and they both behave in the same way. What do I need to do to get the correct bootloader into these devices?

tony1tf, 2022-08-26:
More progress - have got the Arduino SDK on my MacBook Pro communicating with the LilyGO board. Unfortunately, I can't get an example sketch to display on the TFT screen. The sketch 'TFT_ReadWrite_Test runs, but returns zeroes read on the serial monitor, so it looks like I have got to do a further edit on the set-up files to match the board:

08:25:34.969 ->  Pixel value written = 400
08:25:34.969 ->  Pixel value read    = 0
08:25:34.969 ->  ERROR                 ^^^^
08:25:35.477 ->  Pixel value written = 800
08:25:35.477 ->  Pixel value read    = 0
08:25:35.477 ->  ERROR                 ^^^^

LilyGO, 2022-08-26:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup137_LilyGo_TDisplay_RP2040.h

This connection has been verified with no problems. But I haven't tried reading pixels from ST7789.

tony1tf, 2022-08-26:
The 'Display User Setup' sketch shows this:
08:43:56.291 -> [code]
08:43:56.291 -> TFT_eSPI ver = 2.4.72
08:43:56.291 -> Processor    = RP2040
08:43:56.291 -> Transactions = Yes
08:43:56.291 -> Interface    = SPI
08:43:56.291 -> Display driver = 7789
08:43:56.291 -> Display width  = 135
08:43:56.291 -> Display height = 240
08:43:56.291 -> 
08:43:56.291 -> R0 x offset = 52
08:43:56.291 -> R0 y offset = 40
08:43:56.291 -> MOSI    = GPIO 3
08:43:56.291 -> SCK     = GPIO 2
08:43:56.291 -> TFT_CS   = GPIO 5
08:43:56.291 -> TFT_DC   = GPIO 1
08:43:56.291 -> TFT_RST  = GPIO 0
08:43:56.291 -> 
08:43:56.291 -> Font GLCD   loaded
08:43:56.291 -> Font 2      loaded
08:43:56.291 -> Font 4      loaded
08:43:56.291 -> Font 6      loaded
08:43:56.291 -> Font 7      loaded
08:43:56.291 -> Font 8      loaded
08:43:56.291 -> Smooth font enabled
08:43:56.291 -> 
08:43:56.291 -> Display SPI frequency = 40.00
08:43:56.291 -> [/code]

Pietrogramma, 2022-08-27:
Hi, I have the same problem of tony1tf; I followed your indications but I can't resolve, the screen of my Raspberry Pi Pico (purchased on banggood) is always black.
I tried to upload on the board some sketch of the TFT-spi example, but nothing, the screen remain always black.
When I upload a sketch, Arduinio Ide say me:

In file included from C:\Users\Maxi\Documents\Arduino\libraries\TFT_eSPI\examples\Test and diagnostics\TFT_ReadWrite_Test\TFT_ReadWrite_Test.ino:3:
C:\Users\Maxi\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  882 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from C:\Users\Maxi\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
C:\Users\Maxi\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  882 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
Sketch uses 110096 bytes (5%) of program storage space. Maximum is 2093056 bytes.
Global variables use 12108 bytes (4%) of dynamic memory, leaving 250036 bytes for local variables. Maximum is 262144 bytes.
Resetting COM21
Caught exception during reset!
Converting to uf2, output size: 234496, start address: 0x2000
Flashing D: (RPI-RP2)
Wrote 234496 bytes to D:/NEW.UF2

I modified the User_Setup_select.h uncommented "#include <User_Setups/Setup137_LilyGo_TDisplay_RP2040.h>  // Setup file for Lilygo T-Display RP2040 (ST7789 on SPI bus with 135x240 TFT)" but nothing, the screen is always black.

Can you help me, please.

@tony1tf can you send me *.ino file that work correctly with the board, please?

Thanks and with beste regards

LilyGO, 2022-08-27:
Have you tried using this.
https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040/blob/main/example/Arduino/firmware/firmware.ino

Pietrogramma, 2022-08-27:
> Have you tried using this. https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040/blob/main/example/Arduino/firmware/firmware.ino

If I uncomment the line 64 (Serial.available) yes, I can see the screen with  Lilygo logo, with another image and with a written on the voltage

LilyGO, 2022-08-27:
Then please compare the difference between the two in initialization.

Pietrogramma, 2022-08-27:
I take from the TFT_espi example folder the file "Sprite_TF_Rainbow" with this code:
`
#define IWIDTH  160
#define IHEIGHT 128

#include <TFT_eSPI.h> // Graphics and font library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

TFT_eSprite img = TFT_eSprite(&tft);

unsigned long targetTime = 0;
byte red = 31;
byte green = 0;
byte blue = 0;
byte state = 0;
unsigned int colour = red << 11;

void setup(void) {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);

  img.createSprite(IWIDTH, IHEIGHT);
  img.fillSprite(TFT_BLACK);

  targetTime = millis() + 1000;
}

void loop() {

  if (targetTime < millis()) {
    targetTime = millis() + 100;//10000;

    // Colour changing state machine
    for (int i = 0; i < 160; i++) {
      img.drawFastVLine(i, 0, img.height(), colour);
      switch (state) {
        case 0:
          green += 2;
          if (green == 64) {
            green = 63;
            state = 1;
          }
          break;
        case 1:
          red--;
          if (red == 255) {
            red = 0;
            state = 2;
          }
          break;
        case 2:
          blue ++;
          if (blue == 32) {
            blue = 31;
            state = 3;
          }
          break;
        case 3:
          green -= 2;
          if (green == 255) {
            green = 0;
            state = 4;
          }
          break;
        case 4:
          red ++;
          if (red == 32) {
[... 156 lines cut ...]

    //img.drawCentreString("Font size 2",81,12,2); // Draw text centre at position 80, 12 using font 2

    img.drawCentreString("Font size 4", 80, 30, 4); // Draw text centre at position 80, 24 using font 4

    img.drawCentreString("12.34", 80, 54, 6); // Draw text centre at position 80, 24 using font 6

    img.drawCentreString("12.34 is in font size 6", 80, 92, 2); // Draw text centre at position 80, 90 using font 2

    // Note the x position is the top left of the font!

    // draw a floating point number
    float pi = 3.14159; // Value to print
    int precision = 3;  // Number of digits after decimal point
    int xpos = 50;      // x position
    int ypos = 110;     // y position
    int font = 2;       // font number only 2,4,6,7 valid. Font 6 only contains characters [space] 0 1 2 3 4 5 6 7 8 9 0 : a p m
    xpos += img.drawFloat(pi, precision, xpos, ypos, font); // Draw rounded number and return new xpos delta for next print position
    img.drawString(" is pi", xpos, ypos, font); // Continue printing from new x position

    img.pushSprite(0, 0);
  }
}`

But don't work

Pietrogramma, 2022-08-27:
I resolve the iusse, thanks for your support.
in the Void Setup I added
```
pinMode(PWR_ON, OUTPUT);
 digitalWrite(PWR_ON, 1);
```
and now all work fine.
Thanks again

tony1tf, 2022-08-27:
Hi Pietrogramma

I was just working through your example when you said you had fixed it.
I added the following lines to Void Setup:

  pinMode(PWR_ON, OUTPUT); // added for LilyGO
  digitalWrite(PWR_ON, 1); // added for LilyGO
  pinMode(TFT_BL, OUTPUT); // added for LilyGO
  analogWrite(TFT_BL, 150); // added for LilyGO

So I don't have to keep editing the examples, except for Power ON I had also edited the setup137 file:

#define PWR_ON 22. // added by Tony
// Don't uncomment next line if you want PWM control of the backlight in the sketch
#define TFT_BL 4 // uncommented by Tony

I guess LilyGO had forgotten that the setup files need modifying if they want us to conform to Bodmers method for updating the TFT_eSPI library. Plus that Power On command which isn't mentioned anywhere except in the firmware source code.

I read how pleased people are with the LilyGO board in the comments on AliExpress or BangGood, but I am not sure many of them have started to program them yet. 
Try the Animated Eyes test program - it's fascinating. I had to add the Power on code to Setup.
I hope we can all have fun with these boards - eyes in a robot spring to mind!

Tony

Pietrogramma, 2022-08-28:
> Hi Pietrogramma
>
> I was just working through your example when you said you had fixed it. I added the following lines to Void Setup:
>
> pinMode(PWR_ON, OUTPUT); // added for LilyGO digitalWrite(PWR_ON, [...]

Thanks Tony, I'll follow your advice. 
As for the card, I agree with you, yesterday I purchased a ESP32 TTGO board on banggood, also of the Lilygo, which seems to be more performing than the Raspberry, and which also has a WIFi and Blutooth connection.
I will search with google Animated Eyes test program, so I will try it. 
Thanks again and bye

stewbinoh, 2022-11-07:
I sort of have firmware.ino working, but the images are offset on the screen.
If I set rotation to zero, they look better, but I seem to need to add 40 to the 0,0 coordinates, and then there is still screen space that isn't getting used and the images are cropped.

Read_user_setup reports a different display? (I am using the include file mentioned above)

[stew@lenovo TFT_eSPI]$ grep ^#include User_Setup_Select.h
#include <User_Setup.h>           // Default setup is root library folder
#include <User_Setups/Setup137_LilyGo_TDisplay_RP2040.h>  // Setup file for Lilygo T-Display RP2040 (ST7789 on SPI bus with 135x240 TFT)

16:27:59.804 -> [code]
16:27:59.804 -> TFT_eSPI ver = 2.4.72
16:27:59.804 -> Processor    = RP2040
16:27:59.804 -> Transactions = Yes
16:27:59.804 -> Interface    = SPI
16:27:59.804 -> Display driver = 9341
16:27:59.804 -> Display width  = 240
16:27:59.804 -> Display height = 320

deon-dup, 2022-11-28:
Guys

Very desperate. I've worked through this whole thread, and just cannot get my LilyGO RP2040 to display. In summary I've done/have the following:

1. Arduino on Mac OS Catalina
2. Loaded the earlephilhower board as per instruction (Raspberry Pico W)
3. Used the UF port, through USB on MAC. I have no luck with the Bluetooth serial. I do think my ports could be an issue, but I have managed to load the Blink sketch without any issue.
4. Commented out and added the #includes, and other settings as per above
5. Download the sketch, example Sprite, and only get a screen that flashes loads of pixels, nothing else. Tried several sketches under TFT_eSPI, but without any luck.

I'm hoping it is something silly I'm doing wrong, please help.

mmMicky, 2022-11-28:
> Guys
>
> Very desperate. I've worked through this whole thread, and just cannot get my LilyGO RP2040 to display. In summary I've done/have the following:
>
> 1. Arduino on Mac OS Catalina
> 2. Loaded the [...]

Let me assist you with displaying the screen.
1. I will upload a factory firmware, the display should show after you upload (for troubleshooting the display).
[firmware.zip](https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040/files/10101587/firmware.zip)

2. If you use the latest TFT-eSPI library, there will be an independent T-Display-RP2040. Uncomment both to use the routine.
3. Please provide any error information during the process.

deon-dup, 2022-11-28:
Thanks Micky

I’ll try you suggestions and revert later today.

deon-dup, 2022-11-29:
Ok it works with the attached firmware.uf2 file. Now for I guess a really silly question. When I compile any of the example programs, it replaces the file you sent, and I'm back to square 1.

mmMicky, 2022-11-29:
I don't understand you. So now everything is normal and can be compiled and used?

deon-dup, 2022-11-29:
I have compiled (as example TFT_eSPI>Test and diagnostics>Colour_Test), and the UF2 is loaded ot the Pico, which is then unmounted. When I boot, nothing happens. So my understanding is that the UF2 that you sent is replaced by the file I just compiled on the Arduino IDE. This file in my mind has now been replaced by the new UF2, which seemingly does nothing.
 
[image]

mmMicky, 2022-11-29:
I understand what you mean. The .UF2 file I send works fine. But the .UF2 file you compiled is loaded into it and the operation is not normal. The phenomenon shown is that the display screen does not display as expected. I have two suggestions:
1. Upload any serial port information that can be printed, and use the monitor to observe whether it is printed normally. If not, is there a misconfiguration. Screenshot here if necessary. I will help you check.
2. Re-update the latest TFT_eSPI library (https://github.com/Bodmer/TFT_eSPI), and uncomment this line to try
(https://github.com/Bodmer/TFT_eSPI/blob/0fd239c7515cfdecfb64adae8737f27deea96e5a/User_Setup_Select.h#L117).

deon-dup, 2022-11-29:
Hi

I think I'm missing something basic. On point 1 above, I'm using the /dev/cu.usbmodem14201 Serial Port (USB), which does work for Blink. Reloaded Bodmer a few times, and uncommented #include <User_Setups/Setup137_LilyGo_TDisplay_RP2040.h> in User_Setup_Select.h. Now I'm just wondering about my paths. I have 2 locations for the TFT_eSPI, under Documents and Library. Could that be part of the problem?

realdaveblanch, 2023-03-11:
Followed the whole thing, still not working,screen is fully black, uf2 above works, i updated the TFT thing, still black, commented uncommented everything that can be done, now i just have a board that shows lily logo, voltage ,and raspberry logo, none of the examples work, i feel scamed

RoyalB03, 2023-04-18:
I had the same problem with the firmware it worked fine but with the Arduino IDE it failed even though the example upload was successful but the screen was still black, only the blinking light worked. **Can someone help me? - Thanks a lot !**
[image]

realdaveblanch, 2023-04-27:
only way to make it work (as far as i know)

https://steinlaus.de/ttgo-t-display-rp2040-hello-world/

tony1tf, 2023-04-27:
Hi realdaveblanch

Thanks for another link to how to do it.  
This is the extra code - just copied from one of my posts above that got the display working. I added the following lines to Void Setup:

pinMode(PWR_ON, OUTPUT); // added for LilyGO 
digitalWrite(PWR_ON, 1); // added for LilyGO 
pinMode(TFT_BL, OUTPUT); // added for LilyGO 
analogWrite(TFT_BL, 150); // added for LilyGO

So I don't have to keep editing the examples, except for Power ON I had also edited the setup137 file:

#define PWR_ON 22. // added by Tony 
// Don't uncomment next line if you want PWM control of the backlight in the sketch 
#define TFT_BL 4 // uncommented by Tony

Tony

yaricom, 2024-02-22:
For anyone still having a black screen issue - it looks like a conflicting versions of [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) and [arduino-pico](https://github.com/earlephilhower/arduino-pico)

I was able to get screen working by pinpointing the versions of libraries as following:
- [arduino-pico](https://github.com/earlephilhower/arduino-pico) - v3.2.2
- [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) - v2.5.43

After release v3.3.0 of [arduino-pico](https://github.com/earlephilhower/arduino-pico) the screen is blank up to v3.3.2 (which is current at the time of writing this).

jdrowell, 2024-08-15:
I got it to work after git bisecting after 3.3.2, but then found a simple solution here: https://github.com/Bodmer/TFT_eSPI/issues/3402 - just add #define RP2040_PIO_SPI to Setup137_LilyGo_TDisplay_RP2040.h. I didn't test if it affects performance but it's cleaner than my solution, which involves patching pico-sdk (described in the link).

So with that small change one can upgrade to arduino-pico 3.9.5 and not have the black screen problem. You still need to set power on and set the backlight level.