Xinyuan-LilyGO/LILYGO-T-display-RP2040 #14: BOOTSEL compilation error

eyalrad, 2023-01-28:
Hi,
While trying to enable the lvgl example, I get the following compilation error in line
https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040/blob/main/example/Arduino/lvgl/lvgl.ino#L118

Compilation error: 'BOOTSEL' was not declared in this scope

Using the already pre-built binary works perfectly.
Can you kindly help ?
Thanks,
Eyal

eyalrad, 2023-01-28:
Hi,
Issue resolved, after installing Raspberry Pi Pico/RP2040 By Earle F. Philhower, III ( previously I installed Arduino Mbed OS RP2040 Boards By Arduino ).

Current status is that the code is compiled correctly, however the BOOTSEL is not repsonding,
Any help would be appreciated.
Thanks,
Eyal

mmMicky, 2023-01-30:
BOOTSEL should be a parameter defined in the SDK. If it fails to compile, it should be that the header file is missing or this variable has been changed or cancelled.

eyalrad, 2023-01-30:
hi @mmMicky ,
yes, thanks alot.
It worked!
Eyal

closed 2023-01-30 by eyalrad